trichl/TickTagOpenSource #1: Battery runtime

padok, 2022-07-05:
Hi there,

I am very interested in your project, but my use case is a little different. 
I would like to track my animals for a year and wonder what is the average power consumption of the tick tag is if, for example, 3 GPS positions are recorded every day. Very low power setup, as far as I can tell the EEPROM size allows ~35 recordings per day within a year (I would also be interested in data with this setting).
I would like to calculate how many, say, 1500 mAh AA alkaline batteries are needed to run the device for a year.
Do you have any data that could help me figure this out and are you willing to provide it to me?

Thanks in advance!

padok, 2022-07-06:
Thanks for your quick reply!

I mainly want to avoid lipos because I can't make sure that the lipos will operate in their optimal temperature range and their low shelf life is a concern to me. My plan is to connect three AA battery pairs in series, which would give me a voltage range of 2.7V to 4.5V with 6 cells and a capacity of about 15000mAh, which is pretty much what LIPOs deliver at 3.2V to 4.2V. 
Would that work? At what voltage does your firmware shut off? And is 4.5V still in a acceptable range?

From your rough estimation and roughly 3/4 of the battery capacity (4.5V-3.2V) that would mean 30 fixes a day in bad conditions right?

My idea is to maybe replace some TELLUS T3H collars with this setup.

trichl, 2022-07-08:
> Thanks for your quick reply!
>
> I mainly want to avoid lipos because I can't make sure that the lipos will operate in their optimal temperature range and their low shelf life is a concern to me. My [...]

Hi padok, this design is optimized to be ultra light. Such a huge setup is not within the scope of the project and I would recommend using a standard GPS logger off the shelf with a bigger antenna. Any voltage above 4.3V will destroy the logger. The antenna will most probably be detuned by the presence of such a big battery pack. Best