trichl/TickTagOpenSource #3: BOM file no part numbers

ptenax, 2023-01-29:
Hi, 

Thank you for making this available. I would very much like to use it, and I have begun an order with PCBWay. I have used the version 4 files, but I have a message back from PCBWay that the BOM file does not contain part numbers, so I cannot place the order. I submitted BOM-TickTagREV4.xlsx

I also submitted the files: TickTagREV4_2022-01-08.zip and CENTROID_TickTagREV4_2022-01-08.zip

I am super-keen to use this system, but I am not an electronic engineer, so I am hoping you can respond. 

Kyle

trichl, 2023-02-04:
Hey Kyle,

that is strange. Check out the xlsx file, it contains all part numbers, manufacturer ids and so on. We never had an issue with that.

Best, Timm

ptenax, 2023-02-04:
Hi Timm,

I really appreciate your reply.

I checked again by re-downloading the whole zipped repository instead of trying to download just the Excel file. I could open the xlsx from the downloaded zip.  Just my unfamiliarity with Github - I did not realise this would be a problem.

Thanks again. Are you available via your academic email to help with any further questions?

Kyle